linuxserver/docker-orcaslicer #3: Bambu Labs Webcam Crashes OrcaSlicer

fals3illusion, 2024-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Whenever I try to play the webcam, OrcaSlicer crashes

### Expected Behavior

Webcam should show up

### Steps To Reproduce

Go to device, then try to hit play on webcam .  Crashes program after it tries to load the image

### Environment

```markdown
- OS: Debian 12
- How docker service was installed:
Docker Compose on Proxmox Debian LXC with 4GB RAM, 20BGB Storage
```

### CPU architecture

x86-64

### Docker creation

```bash
---
version: "2.1"
services:
  orcaslicer:
    image: lscr.io/linuxserver/orcaslicer:latest
    container_name: orcaslicer
    security_opt:
      - seccomp:unconfined #optional
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Etc/UTC
    volumes:
      - /root/orcaslicer:/config
    ports:
      - 3000:3000
      - 3001:3001
    restart: unless-stopped
```

### Container logs

```bash
Not sure how to find this... new to linuxserver
```

j0nnymoe, 2024-01-08:
Unfortunately no one within the team has a bambu printer so we can't test this. You will need to provide some sort of logs of orcaslicer itself. These can be found in `/config/.orca` I believe (or similar name, but key that it's a hidden folder.)

fals3illusion, 2024-01-08:
[debug_Mon_Jan_08_22_27_36_619.log](https://github.com/linuxserver/docker-orcaslicer/files/13866813/debug_Mon_Jan_08_22_27_36_619.log)

Here is a log file from after it crashes. Hopefully this is some help

LinuxServer-CI, 2024-02-08:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

thelamer, 2024-02-08:
This seems like it might be upstream specific to their Linux version: 
https://github.com/bambulab/BambuStudio/issues/2065

I do this whole song and dance here in an attempt to support the camera: 

https://github.com/linuxserver/docker-bambustudio/blob/master/Dockerfile#L20-L34

And looking further looks to be an issue with mismatched libjpeg libraries: 

https://github.com/bambulab/BambuStudio/issues/1814

I created a sloppy local image using their fedora appimage here: 

```
taisun/random-images:bambutest
```

@fals3illusion  Let me know if this one works for your printer camera and I can work the code into the repo to pull the Fedora appimage instead.

fals3illusion, 2024-02-08:
> This seems like it might be upstream specific to their Linux version: [bambulab/BambuStudio#2065](https://github.com/bambulab/BambuStudio/issues/2065)
> 
> I do this whole song and dance here in an attempt to support the camera:
> 
> https://github.com/linuxserver/docker-bambustudio/blob/master/Dockerfile#L20-L34
> 
> And looking further looks to be an issue with mismatched libjpeg libraries:
> 
> [bambulab/BambuStudio#1814](https://github.com/bambulab/BambuStudio/issues/1814)
> 
> I created a sloppy local image using their fedora appimage here:
> 
> ```
> taisun/random-images:bambutest
> ```
> 
> @fals3illusion Let me know if this one works for your printer camera and I can work the code into the repo to pull the Fedora appimage instead.

I updated my docker image in docker-compose to 

taisun/random-images:bambutest

ran docker-compose up -d

Now I tried to launch the OrcaSlicer App and it is not running.  I get this error.
[image]

thelamer, 2024-02-08:
I actually entered this thinking it was bambustudio. That image is bambustudio. 
You would need to delete the file /config/.config/openbox/autostart ro right click the desktop and open bambu studio. 

Given Orca is a fork I would assume this is the same issue and something you can test for me is if the native linuxserver/bambustudio:latest image works or if this custom taisun/random-images:bambutest is the only one. 

Just don't use a volume mount at all as this is just an ephemeral test and try those two out. That way we will know if this orca image and bambu are suffering from the same lib mismatch, it really helps narrow down the troubleshooting as I have no test setup to confirm.

fals3illusion, 2024-02-08:
> taisun/random-images:bambutest
I have an instance of the bambustudios running in the same way as orcaslicer.  

I just checked before the custom image pull and it did not work.  Using the custom image you provided, the Bambu one now works !  I am getting the webcam to work without crashing the app. 

I tried to delete the autostart file and still got the error when right clicking the desktop and trying to open orcsslicer.  

[image]

thelamer, 2024-02-08:
This is really good news as I now have a comparison and we know specifically what the issue is. The fedora version of the appimage bundles the libs it needs and it is this lib mismatch so I will try pointing this out upstream and see if they can get a fix in. I will also try to manually fix the libs in the image, but I will not have any time until this weekend. 
The fedora ingestion is easy so I will get that into the Bambu Studio image before then. 

Thanks for testing  I will update this thread when I have something.

fals3illusion, 2024-02-08:
> This is really good news as I now have a comparison and we know specifically what the issue is. The fedora version of the appimage bundles the libs it needs and it is this lib mismatch so I will try [...]

No problem, thanks for you help !  
It's just easier to remotely manage my slicer/printer on a webpage anywhere  vs waking my pc and then RDP into that machine and then slice/upload/ check the camera lol

Pastitas, 2024-03-16:
Is this also fixable for orcaslicer version?

dohlin, 2024-04-23:
@thelamer are there any updates that you know of on this by chance?  I see there is a merge above for bambustudio, but I'm still having the same crashing issue with the latest orcaslicer docker build.  Thank you!

j0nnymoe, 2024-04-23:
Are you able to provide any error logs for when the crash happens? No one within the team has a bambu printer so can't test.

dohlin, 2024-04-24:
> Are you able to provide any error logs for when the crash happens? No one within the team has a bambu printer so can't test.

I apologize, I assumed it was the same issue.  I've tried to pull the log files, but the only ones I can find are in /.config/OrcaSlicer/log.

And unfortunately, the files that look like they might be valuable (more than 0KB) end in .log.enc and when opened in a text editor result in gibberish.  Am I missing something here?
[image]

bingo-i, 2024-06-05:
I am also experiencing the same problem.
It works fine in linuxserver/bambustudio, but
In orca, the default webcam of the Bamboo Lab printer does not work and the program is forcefully terminated or generates an error code.

nakermann1973, 2024-09-30:
I have another +1

The log that AppRun outputs on stdout just before crashing is:

```
BambuTunnel::GetMsg(1)
Segmentation fault
```

The debug log contents are as follows.  the log.enc files were 0-sized:
```
chris@snark ~ $ cat /tmp/debug_Mon_Sep_30_18_13_10_1190.log.0 
[warning]	2024-09-30 18:13:11.163906[Thread 0x00007f5b57e59440]:get_network_function, can not find function bambu_network_del_subscribe
[warning]	2024-09-30 18:13:11.164012[Thread 0x00007f5b57e59440]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2024-09-30 18:13:11.164031[Thread 0x00007f5b57e59440]:get_network_function, can not find function bambu_network_get_profile_3mf
[warning]	2024-09-30 18:13:12.721300[Thread 0x00007f5b57e59440]:06:13:12 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-09-30 18:13:12.723951[Thread 0x00007f5b57e59440]:06:13:12 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-09-30 18:13:12.727781[Thread 0x00007f5b57e59440]:06:13:12 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-09-30 18:13:12.766700[Thread 0x00007f5b57e59440]:06:13:12 PM: Warning: Error running JavaScript: Unsupported result type
```

closed 2025-09-18 by j0nnymoe