linuxserver/docker-orcaslicer #6: Google Login - failes

ejg3855, 2024-04-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When I click the login with google account it takes me to the page, I login, and authenticate with my mobile device then the redirect page from bambulabs just spins endlessly

### Expected Behavior

login to work properly

### Steps To Reproduce

install orca slicer
login to bambu account with network plugin

### Environment

```markdown
- OS:
- How docker service was installed: through unraid apps store
```

### CPU architecture

x86-64

### Docker creation

```bash
unsure
```

### Container logs

```bash
not sure how to provide this
```

aptalca, 2024-04-03:
Please google `unraid docker faq` and it should tell you how to get the `docker run` and the `container logs` that we are asking for in the template

christophtoelle, 2024-04-25:
I got the same issue, copying the link and moving it into incognito-tab did the trick for me

LinuxServer-CI, 2024-05-25:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

closed 2024-07-24 by LinuxServer-CI