linuxserver/docker-orcaslicer #8: [BUG] Error when sending G-code files over LAN

MorganMLGman, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

Orcaslicer displays error 0 when using **SEND** option, instead of **PRINT**.

[image]

### Expected Behavior

File is sent successfully.

### Steps To Reproduce

1. Run Orcaslicer
2. Login with Bambu account
3. Slice some models
4. Send gcode to printer

### Environment

```markdown
- OS: Debian 12
- How docker service was installed: distro's packagemanager
```

### CPU architecture

x86-64

### Docker creation

```bash
orcaslicer:
    image: linuxserver/orcaslicer:latest
    container_name: orcaslicer
    restart: unless-stopped
    security_opt:
      - seccomp:unconfined #optional
    ports:
      - 3005:3005
      - 3006:3006
    volumes:
      - $DOCKER_APPS_DIR/orcaslicer:/config
      - $DOCKER_APPS_DIR/orcaslicer/models:/models
    devices:
      - /dev/dri:/dev/dri
    environment:
      - PUID=$PUID
      - PGID=$PGID
      - TZ=$TZ
      - CUSTOM_PORT=3005
      - CUSTOM_HTTPS_PORT=3006
      - TITLE=Orca Slicer
      - FM_HOME=/models
```

### Container logs

```bash
https://gist.github.com/MorganMLGman/8e23077e5c9af9dc0377d18c3d951653
```

LinuxServer-CI, 2024-07-07:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

MorganMLGman, 2024-07-07:
Issue still valid

j0nnymoe, 2024-07-07:
If you can provide some logs from orca itself, that would be helpful. No one within the team have a bambu printer so not currently able to replicate.

MorganMLGman, 2024-07-07:
Will provide as soon as possible

MorganMLGman, 2024-07-14:
`
[warning]       2024-07-14 14:49:41.240835[Thread 0x00007f38f5348440]:get_network_function, can not find function bambu_network_del_subscribe
[warning]       2024-07-14 14:49:41.240957[Thread 0x00007f38f5348440]:get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]       2024-07-14 14:49:41.240981[Thread 0x00007f38f5348440]:get_network_function, can not find function bambu_network_get_profile_3mf
[error] 2024-07-14 14:49:41.880708[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:41.891691[Thread 0x00007f383c7f86c0]:Unable to open the file/models/Dog Tag/DogTag.step
[error] 2024-07-14 14:49:41.892704[Thread 0x00007f1fb9ffb6c0]:Unable to open the file/models/Dron/JeNo_3D_VFX_GPS_Mount.step
[warning]       2024-07-14 14:49:42.079603[Thread 0x00007f38f5348440]:02:49:42 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-07-14 14:49:42.092907[Thread 0x00007f38f5348440]:02:49:42 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-07-14 14:49:42.095103[Thread 0x00007f38f5348440]:02:49:42 PM: Warning: Error running JavaScript: Unsupported result type
[warning]       2024-07-14 14:49:42.113581[Thread 0x00007f38f5348440]:02:49:42 PM: Warning: Error running JavaScript: Unsupported result type
[error] 2024-07-14 14:49:43.299962[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:43.335844[Thread 0x00007f38f5348440]:connect_server error: network_agent=0x55c2634a4910, ret=-2
[warning]       2024-07-14 14:49:43.726412[Thread 0x00007f1f40dfb6c0]:[Orca Updater]: online version=1.9.3.1, current_version=1.9.3.1, no need to download
[error] 2024-07-14 14:49:53.706223[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:53.968237[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:53.983740[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:53.998901[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.015370[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.032214[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.032941[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.033477[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.049685[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:49:54.278514[Thread 0x00007f38f5348440]:tree_sel_change_delayedcan not find page with current selection
[error] 2024-07-14 14:50:06.018432[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.019068[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.019616[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.020127[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.020625[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.021108[Thread 0x00007f38f5348440]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-07-14 14:50:06.043576[Thread 0x00007f38f5348440]:tree_sel_change_delayedcan not find page with current selection
[error] 2024-07-14 14:50:06.074474[Thread 0x00007f38f5348440]:tree_sel_change_delayedcan not find page with current selection
[error] 2024-07-14 14:50:09.696247[Thread 0x00007f38f5348440]:tree_sel_change_delayedcan not find page with current selection
`

MorganMLGman, 2024-07-14:
I don't know what's going on with the `[error] Unable to open file...` as file I sliced has nothing to do with these missing files.
Also, I don't know what's going on with `[error] Unable to create exclude triangles` as file was first sliced with *Slice plate* button without any warnings and errors in UI, and then I used *Send plate*

guilhermeteruaki, 2024-10-07:
Hi, I'm having the same problem

lockheed, 2024-12-07:
Again, same issue here.

j0nnymoe, 2024-12-08:
Everyone here with the same issue using a Bambu printer like OP? or something different?

lockheed, 2024-12-08:
Bamby P1S + AMS

rainabba, 2024-12-14:
I'm on the X1C + AMS. Also, the Orcaslicer logo on the login panel is extremely counter-intuitive for those of us trying to use Orcaslicer with our Bambu printers.

j0nnymoe, 2024-12-14:
Unfortunately no one within the team has a bambu printer so it's not something we're able to test. I assume the logging into an account works ok?

guilhermeteruaki, 2024-12-14:
Login works normally.
I can send a plate to print normally, but if I try to send to the microsd I get the error

rainabba, 2024-12-14:
In case it's relevant, the LAN Only Access Code is a 8-character, alpha-numeric value, not a 6-digit pin. I wonder if Bambu might have changed access in a way that LAN Only mode isn't working as before?

j0nnymoe, 2024-12-14:
One test would be to try the native install of orca.

exen904, 2024-12-22:
Native makes no difference, had the same issue with either the AppImage and right now also with the Windows version. Tries to send either through Cloud Service, switches to LAN in the second try and also fails. Browsing the SDHC is working just fine, using Print and not Send also works just fine.

- Orca 2.2.0
- Windows
- A1 with AMS

j0nnymoe, 2024-12-22:
Thanks for confirming it's not a container issue. Best to be reported upstream.

closed 2025-03-22 by LinuxServer-CI

LinuxServer-CI, 2025-04-21:
This issue is locked due to inactivity