linuxserver/docker-orcaslicer #9: [FEAT] Dark mode missing

aurarius1, 2024-07-07:
### Is this a new feature request?

- [X] I have searched the existing issues

### Wanted change

OrcaSlicer on desktop has a dark mode, is this feature disabled for the container?

### Reason for change

Because OrcaSlicer on desktop has it.

LinuxServer-CI, 2024-08-06:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

XxSimpleAceXx, 2024-09-22:
+1 darkmode would be nice to see implemented

arraylabs, 2024-11-02:
This would be great!

machetie, 2024-11-07:
please add this feature, the only thing that stops me using it in docker.

ss1gohan13, 2024-12-11:
+1 for dark mode

j0nnymoe, 2024-12-11:
https://github.com/SoftFever/OrcaSlicer/issues/883

It doesn't seem to be an option within the appimage version of OrcaSlicer, seems it has to be modified via the DE settings.

ss1gohan13, 2024-12-11:
> [SoftFever/OrcaSlicer#883](https://github.com/SoftFever/OrcaSlicer/issues/883)
> 
> It doesn't seem to be an option within the appimage version of OrcaSlicer, seems it has to be modified via the DE settings.

Could you elaborate into this a little bit for me?

AndreaPro, 2024-12-21:
> please add this feature, the only thing that stops me using it in docker.

Same!

j0nnymoe, 2024-12-21:
> > [SoftFever/OrcaSlicer#883](https://github.com/SoftFever/OrcaSlicer/issues/883)
> > It doesn't seem to be an option within the appimage version of OrcaSlicer, seems it has to be modified via the DE settings.
> 
> Could you elaborate into this a little bit for me?

Not really, it's been mentioned within that thread that it isn't part of their app image release.

nsyring, 2024-12-27:
This is a great Image. I also would appreciate darkmode. I think this has to be done in the openbox-settings paired with some theme-package installs.

j0nnymoe, 2024-12-27:
The current solution is janky at best. This should just be added upstream.

wernight, 2025-01-25:
To use dark themed, set `-e GTK_THEME=Adwaita:dark`  or the equivalent environment in `docker-compose.yml`.

AndreaPro, 2025-01-25:
> To use dark themed, set `-e GTK_THEME=Adwaita:dark` or the equivalent environment in `docker-compose.yml`.

It works!

closed 2025-04-27 by LinuxServer-CI