linuxserver/docker-orcaslicer #10: [FEAT] upload files to slicer

street-grease-coder, 2024-08-21:
### Is this a new feature request?

- [X] I have searched the existing issues

### Wanted change

It's an amazing project, thank you a ton. It would be great if files could be selected from the browser, and uploaded to the slicer / destination pc. It might be a bit of hassle, but it would be amazing in terms of convenience 

### Reason for change

convenience, no need to upload files to host pc anymore

### Proposed code change

redirect file selection to vnc file uploader, and select uploaded file to import

aptalca, 2024-08-21:
At the top middle, there should be a file cabinet icon that becomes visible when you open the side menu. That icon opens a file browser and allows you to upload files into the container filesystem through the browser.

LinuxServer-CI, 2024-09-20:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

closed 2024-11-20 by LinuxServer-CI