linuxserver/docker-orcaslicer #11: [FEAT] Variable to disable ipv6

broizter, 2024-08-22:
### Is this a new feature request?

- [X] I have searched the existing issues

### Wanted change

I wan't an environment variable to disable IPV6 in nginx.

### Reason for change

If you have ipv6 disabled on the host the container won't start because nginx tries to bind to ipv4 and ipv6.

broizter, 2024-08-22:
It's possible to bind a nginx configuration file to override this behaviour so it's possible to workaround it although it would be better to set it at container creation via a environment variable.

Roxedus, 2024-08-22:
You can use the unsupported (and thus undocumented) variable `DISABLE_IPV6` to disable ipv6

closed 2024-08-22 by Roxedus