linuxserver/docker-orcaslicer #12: [BUG] Container Crash on Screen Change

bobbieluke1, 2024-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

On changing of screen or pop-up the container can crash and will then only show a completely black screen until container is restarted

### Expected Behavior

Not crash

### Steps To Reproduce

Inconsistent behaviour but always on a page/screen change like trying to edit Printer Presents or within that tab Machine G-Code.

### Environment

```markdown
- OS: Unraid (Slackware 15.0 x86_64)
- How docker service was installed: via Unraid package manager then Docker Compose addon
```

### CPU architecture

x86-64

### Docker creation

```bash
Docker Compose on Unraid:

services:
########################################################################
########################################################################
  orcaslicer:
    image: lscr.io/linuxserver/orcaslicer:latest
    container_name: orcaslicer
    security_opt:
      - seccomp:unconfined #optional
    volumes:
      - "/mnt/user/appdata/containers/orcaslicer/config:/config"
      - "/mnt/user/Projects/3D Printing/Models/Prints:/prints"
      - "/mnt/user/Projects/3D Printing/Models/Downloads:/downloads"
    ports:
      - 3033:3033
      - 3032:3032
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Europe/London
      - CUSTOM_PORT=3033
      - CUSTOM_HTTPS_PORT=3032
      - TITLE=Orca Slicer #Tab title
#      - NO_DECOR=true #If set the application will run without window borders for use as a PWA
    labels:
      - autoheal-app=true
      - com.centurylinklabs.watchtower.enable=true
    healthcheck:
      test: curl --fail http://192.168.0.5:3033/ || exit 1
      interval: 10s
      timeout: 10s
      retries: 3
      start_period: 5s
    restart: unless-stopped
########################################################################
```

### Container logs

```bash
(orca-slicer:267): Gtk-CRITICAL **: 13:58:48.680: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:267): Gtk-CRITICAL **: 13:58:48.680: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:267): Gtk-WARNING **: 13:58:48.693: Negative content width -2 (allocation 18, extents 10x10) while allocating gadget (node button, owner GtkToggleButton)
```

LinuxServer-CI, 2024-10-03:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

kferrerodktire, 2024-10-27:
I'm having the same issue

MRamon550, 2025-01-08:
Same issue as the others

MRamon550, 2025-01-08:
It looks like this keeps getting marked as stale without activity.  What is needed to help troubleshoot?  OP already posted some logs so happy to provide anything else needed

closed 2025-04-10 by LinuxServer-CI

LinuxServer-CI, 2025-05-10:
This issue is locked due to inactivity