linuxserver/docker-orcaslicer #13: [BUG] ERROR: openbox-xdg-autostart requires PyXDG to be installed

blackadar, 2024-10-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

When launching even without any existing config, the container does not launch Orcaslicer although it does display a black screen with the Kasm overlay in browser. The logs show the following error:

```
orcaslicer  | (orca-slicer:305): dbind-WARNING **: 01:36:56.850: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.280: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.280: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.281: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.281: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.283: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.283: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.284: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.284: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.285: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.285: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.286: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.286: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.288: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.288: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.289: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.289: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.290: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.290: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:305): GLib-GObject-WARNING **: 01:36:57.291: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:305): Gtk-CRITICAL **: 01:36:57.291: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  | Aborted (core dumped)
orcaslicer  |
orcaslicer  | ERROR: openbox-xdg-autostart requires PyXDG to be installed
```

### Steps To Reproduce

Launch container

### Environment

```markdown
- OS: Ubuntu 24.04.1
- How docker service was installed: Docker official repo
```

### CPU architecture

x86-64

### Docker creation

```bash
services:
  orcaslicer:
    image: lscr.io/linuxserver/orcaslicer:latest
    container_name: orcaslicer
    security_opt:
      - seccomp:unconfined #optional
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Etc/UTC
      - DRINODE=/dev/dri/renderD128
      - CUSTOM_USER=jordan
      - PASSWORD=$ORCA_PASSWORD
    volumes:
      - ./orca:/config
    devices:
      - /dev/dri/renderD128:/dev/dri/renderD128
    ports:
      - 3002:3000
      - 3001:3001
    restart: unless-stopped
```

### Container logs

```bash
Provided above
```

blackadar, 2024-10-27:
For additional info, rolling back to v2.1.1-ls51 resolved the issue for me

semtex1987, 2024-10-28:
> For additional info, rolling back to v2.1.1-ls51 resolved the issue for me

+1

Same issue here, though rolling back to v2.1.1-ls51 did not  fix it for me.

`2024-10-28 01:46:46,196 [INFO] websocket 33: got client connection from 172.86.12.11
 2024-10-28 01:46:46,205 [DEBUG] websocket 33: using SSL socket
 2024-10-28 01:46:46,205 [DEBUG] websocket 33: X-Forwarded-For ip 'XXXX'
 2024-10-28 01:46:46,210 [DEBUG] websocket 33: BasicAuth matched
 2024-10-28 01:46:46,210 [DEBUG] websocket 33: using protocol Hixie 75
 2024-10-28 01:46:46,210 [DEBUG] websocket 33: connecting to VNC target
 2024-10-28 01:46:46,210 [DEBUG] XserverDesktop: new client, sock 35
 2024-10-28 01:46:46,227 [INFO] EncodeManager: WEBP benchmark result: 10 ms
 2024-10-28 01:46:46,227 [PRIO] Connections: accepted: XXXXX@XXXXX_1730098006.210469::websocket
 2024-10-28 01:46:46,227 [DEBUG] XserverDesktop: Got request for framebuffer resize to 1680x844
 2024-10-28 01:46:46,227 [DEBUG] XserverDesktop: 1 screen(s)
             0 (0x00000000): 1680x844+0+0 (flags 0x00000000)

 2024-10-28 01:46:46,227 [DEBUG] RandR: Resizing screen framebuffer to 1680x844
 2024-10-28 01:46:46,233 [INFO] ComparingUpdateTracker: 762.88 kpixels in / 762.88 kpixels out
 2024-10-28 01:46:46,233 [INFO] ComparingUpdateTracker: (1:1 ratio)
 2024-10-28 01:46:46,234 [DEBUG] RandR: Reconfiguring new output 'VNC-0' to 1680x844+0+0
 2024-10-28 01:46:46,234 [DEBUG] Config: set MaxVideoResolution(String) to 960x540
 2024-10-28 01:46:46,234 [DEBUG] SMsgReader: Client sent config param maxVideoResolution 960x540, applied

(orca-slicer:149): Gtk-CRITICAL **: 01:46:46.432: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:149): Gtk-CRITICAL **: 01:46:46.432: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:149): Gtk-CRITICAL **: 01:46:46.432: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.686: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.686: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.687: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.687: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.697: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.697: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.697: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.697: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.704: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.704: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.709: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.709: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:149): GLib-GObject-WARNING **: 01:46:48.710: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:149): Gtk-CRITICAL **: 01:46:48.710: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
 2024-10-28 01:46:51,778 [DEBUG] SMsgReader: Received binary clipboard, type text/plain, 27 bytes
 2024-10-28 01:46:51,778 [DEBUG] Selection: Remote binary clipboard announced, grabbing local ownership
 2024-10-28 01:46:51,778 [DEBUG] Selection: Created selection window
 2024-10-28 01:46:51,778 [DEBUG] Selection: Grabbed PRIMARY selection
 2024-10-28 01:46:51,778 [DEBUG] Selection: Grabbed CLIPBOARD selection
 2024-10-28 01:46:51,778 [DEBUG] Selection: Client sent 1 mimes, 0 were valid
New WebSocket Connection:  ::ffff:172.86.12.11 Go-http-client/1.1 (1 total)
 2024-10-28 01:46:52,147 [INFO] EncodeManager: Max encoding time during the last 24 frames: 5 ms (limit 41, near limit 33)
Segmentation fault (core dumped)

ERROR: openbox-xdg-autostart requires PyXDG to be installed
`

RiDDiX, 2024-10-31:
Easy fix. Clear within your "mounted" Folder everything except: /.config/OrcaSlicer

Because there are stored your OrcaSlicer Configurations.

LinuxServer-CI, 2024-12-01:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

Schuermi7, 2024-12-08:
Having the same Problem, tried different versions but it keeps happening.

Schuermi7, 2024-12-08:
Having the same behavior, KASM is a black screen and the attempt to start the Slicer returns an error message similar to OPs.

`orcaslicer  | (orca-slicer:267): dbind-WARNING **: 20:47:21.768: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.959: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.959: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.961: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.961: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.962: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.962: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.963: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.963: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.965: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.965: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.966: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.966: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.967: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.967: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.968: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.968: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.970: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.971: invalid cast from 'wxPizza' to 'GtkCellLayout'
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.971: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
orcaslicer  | 17
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.067: gtk_window_resize: assertion 'height > 0' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.081: gtk_window_resize: assertion 'height > 0' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.083: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.083: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.175: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 196 lines cut ...]
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.128: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.138: gtk_window_resize: assertion 'height > 0' failed
orcaslicer  |
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.138: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  |
[... cut ...]
orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.493: gtk_widget_set_size_request: assertion 'width >= -1' failed
orcaslicer  | LLVM ERROR: 64-bit code requested on a subtarget that doesn't support it!
orcaslicer  | Aborted (core dumped)
orcaslicer  |
orcaslicer  | ERROR: openbox-xdg-autostart requires PyXDG to be installed
orcaslicer  |  2024-12-08 20:47:28,025 [INFO] websocket 0: got client connection from 127.0.0.1
orcaslicer  |  2024-12-08 20:47:28,037 [PRIO] Connections: accepted: @91.67.117.67_1733687248.25650::websocket
`

RiDDiX, 2024-12-08:
> Having the same behavior, KASM is a black screen and the attempt to start the Slicer returns an error message similar to OPs.
> 
> OS is Debian 11
> 
> ```
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.959: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.959: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.961: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.961: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.962: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.962: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.963: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.963: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.965: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.965: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.966: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.966: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.967: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.967: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.968: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.968: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.970: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): GLib-GObject-WARNING **: 20:47:21.971: invalid cast from 'wxPizza' to 'GtkCellLayout'
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:21.971: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> orcaslicer  | 17
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.067: gtk_window_resize: assertion 'height > 0' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.071: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.081: gtk_window_resize: assertion 'height > 0' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.083: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:22.083: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 202 lines cut ...]
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.138: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.138: gtk_widget_set_size_request: assertion 'width >= -1' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.139: gtk_window_resize: assertion 'height > 0' failed
> orcaslicer  |
> orcaslicer  | (orca-slicer:267): Gtk-CRITICAL **: 20:47:23.152: gtk_window_resize: assertion 'height > 0' failed
> orcaslicer  |
[... cut ...]
> orcaslicer  | LLVM ERROR: 64-bit code requested on a subtarget that doesn't support it!
> orcaslicer  | Aborted (core dumped)
> orcaslicer  |
> orcaslicer  | ERROR: openbox-xdg-autostart requires PyXDG to be installed
> orcaslicer  |  2024-12-08 20:47:28,025 [INFO] websocket 0: got client connection from 127.0.0.1
> orcaslicer  |  2024-12-08 20:47:28,037 [PRIO] Connections: accepted: @91.67.117.67_1733687248.25650::websocket
> ```

Easy fix. Clear within your "mounted" Folder everything except: /.config/OrcaSlicer

Because there are stored your OrcaSlicer Configurations.

Schuermi7, 2024-12-08:
I deleted as you referred a couple of times and that didn't change a thing.
Removing everything from the folder didn't help either.
Clearing the whole project and volumes didn't work as well

j0nnymoe, 2024-12-08:
> I deleted as you referred a couple of times and that didn't change a thing. Removing everything from the folder didn't help either. Clearing the whole project and volumes didn't work as well

Are you doing a fresh deployment?

Schuermi7, 2024-12-08:
Yeah, I did a fresh deployment of everything

j0nnymoe, 2024-12-08:
ok, so your issue isn't the same as OP. Please open a fresh issue and provide all the requested information.

Schuermi7, 2024-12-08:
As mentioned by OP, he is also without any existing config and thus a fresh deployment, so I would argue, this is the same issue.

j0nnymoe, 2024-12-08:
Ok, still, please provide information of your setup, our CI tests show it boots fine.

Schuermi7, 2024-12-08:
My setup is a Debian 11 setup with Docker (non-Swarm), using the provided docker-compose.yml example from the official documentation.
The only difference is, that there is a reverse proxy in front of it, but that doesn't affect the container.
Tested tags are latest, 2.1.1 and 2.2.0-ls58. Always starting on a fresh install.

j0nnymoe, 2024-12-08:
What's the hardware? What's your docker compose? Are you able to deploy any of our other kasmvnc based containers without problem? Or does the same thing happen?

Schuermi7, 2024-12-08:
My docker-compose.yml looks as follows:
```---
services:
  orcaslicer:
    image: lscr.io/linuxserver/orcaslicer:latest
    container_name: orcaslicer
    security_opt:
      - seccomp:unconfined #optional
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Europe/Berlin
    volumes:
      - ./config:/config
#    ports:
#      - 3000:3000
#      - 3001:3001
    restart: unless-stopped
    labels:
      - traefik.enable=true
      - traefik.http.routers.orca.rule=Host(`orca.foo.rocks`)
      - traefik.http.routers.orca.tls.certresolver=letsencrypt
      - traefik.http.services.orca.loadbalancer.server.port=3000
```

The hardware is a KVM-based virtual server with Intel-hardware. no GPU-acceleration. 

Temporarily deployed "docker-spotube" project and it fails to open up the spotube application as well. This time with: 

```spotube  | (spotube:394): dbind-WARNING **: 22:00:55.399: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
spotube  |
spotube  | (spotube:394): Atk-CRITICAL **: 22:00:55.504: atk_socket_embed: assertion 'plug_id != NULL' failed
spotube  | package:media_kit_libs_linux registered.
spotube  | LLVM ERROR: 64-bit code requested on a subtarget that doesn't support it!
```

j0nnymoe, 2024-12-08:
That LLVM error seems telling, what file system/how is your drive setup?

Schuermi7, 2024-12-08:
The folder is on my main drive, running an ext4 filesystem.
The path is in my root-Homedirectory /root/orcaslicer/config. File Permissions are set to user 1000, group 1000 and all files are written and read correctly.

j0nnymoe, 2024-12-08:
Please try it from the home dir `/home/username/...` and see if that changes anything.

Schuermi7, 2024-12-08:
Added a fresh user "orcaslicer", created the compose-file in the /home/orcaslicer directory, started the project and got the same error as before. 

Adding the output of `lscpu` for added measure.

```Architecture:                       x86_64
CPU op-mode(s):                     32-bit, 64-bit
Byte Order:                         Little Endian
Address sizes:                      40 bits physical, 48 bits virtual
CPU(s):                             6
On-line CPU(s) list:                0-5
Thread(s) per core:                 1
Core(s) per socket:                 6
Socket(s):                          1
NUMA node(s):                       1
Vendor ID:                          AuthenticAMD
CPU family:                         6
Model:                              13
Model name:                         QEMU Virtual CPU version 2.5+
Stepping:                           3
CPU MHz:                            1996.249
BogoMIPS:                           3992.49
Hypervisor vendor:                  KVM
Virtualization type:                full
L1d cache:                          384 KiB
L1i cache:                          384 KiB
L2 cache:                           3 MiB
L3 cache:                           16 MiB
NUMA node0 CPU(s):                  0-5
Vulnerability Gather data sampling: Not affected
Vulnerability Itlb multihit:        Not affected
Vulnerability L1tf:                 Not affected
Vulnerability Mds:                  Not affected
Vulnerability Meltdown:             Not affected
Vulnerability Mmio stale data:      Not affected
Vulnerability Retbleed:             Not affected
Vulnerability Spec rstack overflow: Not affected
Vulnerability Spec store bypass:    Mitigation; Speculative Store Bypass disabled via prctl and seccomp
Vulnerability Spectre v1:           Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:           Mitigation; Retpolines, IBPB conditional, STIBP disabled, RSB filling, PBRSB-eI
                                    BRS Not affected
Vulnerability Srbds:                Not affected
Vulnerability Tsx async abort:      Not affected
Flags:                              fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx
                                     fxsr sse sse2 ht syscall nx lm nopl cpuid tsc_known_freq pni pclmulqdq ssse3 c
                                    x16 sse4_1 sse4_2 x2apic popcnt aes xsave rdrand hypervisor lahf_lm cmp_legacy
                                    abm 3dnowprefetch ssbd ibpb vmmcall arch_capabilities```

j0nnymoe, 2024-12-08:
You didn't need to create a new user, just wanted you to try with the user that had id 1000.

What CPU do you have?

Schuermi7, 2024-12-08:
I am customer with a vServer hoster, that uses AMDs Epyc Server Processors.

And it also doesn't work with the User on ID 1000

j0nnymoe, 2024-12-08:
Ok, at this point, I'm lost for ideas why you're having this issue. I don't believe this to be a container issue but it's more either how your host is setup or (this is a long shot) some CPU feature that's missing. Will need to wait for someone else to join in.

Though as I suspected the OP managed to boot the container and it just failed on upgrades. Your issue does not seem to be the same.

Schuermi7, 2024-12-08:
I would be interested in a full-log from OP to see if they also see the `orcaslicer  | LLVM ERROR: 64-bit code requested on a subtarget that doesn't support it!` line.

But yeah, I guess with what we discussed and what my research on the MESA/LLVM side brought, it could very well be a CPU-identification mismatch produced by the LLVM/MESA subsystems that are present in the KMSAVNC base-image.

Schuermi7, 2025-01-10:
As per my last suspicion, I could indeed confirm, that the problem I had was due to a faulty/not exactly correct KVM configuration on the side of hosters. 

`
Architecture:                         x86_64
CPU op-mode(s):                       32-bit, 64-bit
Byte Order:                           Little Endian
Address sizes:                        40 bits physical, 48 bits virtual
CPU(s):                               6
On-line CPU(s) list:                  0-5
Thread(s) per core:                   1
Core(s) per socket:                   6
Socket(s):                            1
NUMA node(s):                         1
Vendor ID:                            AuthenticAMD
**CPU family:                           15
Model:                                107**
Model name:                           QEMU Virtual CPU version 2.5+
Stepping:                             1
CPU MHz:                              1996.250
BogoMIPS:                             3992.50
Hypervisor vendor:                    KVM
Virtualization type:                  full
L1d cache:                            384 KiB
L1i cache:                            384 KiB
L2 cache:                             3 MiB
L3 cache:                             16 MiB
NUMA node0 CPU(s):                    0-5
Vulnerability Gather data sampling:   Not affected
Vulnerability Itlb multihit:          Not affected
Vulnerability L1tf:                   Not affected
Vulnerability Mds:                    Not affected
Vulnerability Meltdown:               Not affected
Vulnerability Mmio stale data:        Not affected
Vulnerability Reg file data sampling: Not affected
Vulnerability Retbleed:               Not affected
Vulnerability Spec rstack overflow:   Not affected
Vulnerability Spec store bypass:      Not affected
Vulnerability Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointer sanitization
Vulnerability Spectre v2:             Mitigation; Retpolines, IBPB conditional, STIBP disabled, RSB filling, PBRSB-eIBRS Not affected
Vulnerability Srbds:                  Not affected
Vulnerability Tsx async abort:        Not affected
Flags:                                fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm rep_good nopl cpuid extd_apicid tsc_known_freq pni pclmulqdq ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes xsave avx rdrand hyper
                                      visor lahf_lm cmp_legacy abm 3dnowprefetch ssbd ibpb vmmcall arch_capabilities
`

blackadar, 2025-02-10:
Still facing this issue, and nuking everything but .config did not fix it. Later image versions have not fixed the issue.

j0nnymoe, 2025-02-10:
Like @Schuermi7 had issues running this on a vps/VM - are you perhaps doing the same?

blackadar, 2025-02-11:
No this is on Ubuntu 24.04.1 LTS on bare metal, on an Intel i3-N300 with integrated graphics.

vajonam, 2025-02-23:
I have built my own, I tried to switch over to linuxserver (one less thing for me maintain), but did work. you can try this if it works for you. https://github.com/vajonam/orcaslicer-novnc. Its specific to NVIDIA hardware.

vajonam, 2025-02-24:
@j0nnymoe 

I notice when it docker exec a shell into the container and run /opt/orcaslicer/AppRun, it doesn't segfault or crash. 

I am running on beremetal, with other nvida runitme docker containers working fine. I have rolled my own which also works, the autostart in this container doesn't seem to work.

vajonam, 2025-02-24:
running its a `root` works, but not as `abc`

vajonam, 2025-02-24:
`sudo -u abc GTK_THEME=Adwaita:dark /opt/orcaslicer/AppRun` in the autorun file seems to help it launch.. something wierd about environments or perms..

blackadar, 2025-02-24:
Hmm I am using an Intel integrated GPU otherwise I'd give your image a try.

When you say autorun file, do you mean https://github.com/linuxserver/docker-orcaslicer/blob/master/root/defaults/autostart? So this would need a container build to change, or is it on the filesystem somewhere?

On a side note.. does that GTK_THEME give Orca a "dark mode" in this container? If so I wonder if an environment variable for it could solve https://github.com/linuxserver/docker-orcaslicer/issues/9 👀

vajonam, 2025-02-24:
@blackadar  it ends up up inside the `/config/.config/openbox/autostart` you can just edit it there. Yes the environment variable modifies the dark mode.

vajonam, 2025-02-24:
So anyhow, with NVIDIA proprietary drivers, one must launch the tool with virtualgl or “vglrun,” without which the slicer is not very usable. This works but, like I said, since I already maintain one image, there’s no point in adding to this one until it has this functionality, which it probably won’t, since DRI3 support is how KSVM is looking to add GPU acceleration to 3D rendering.

blackadar, 2025-02-25:
Got the same thing unfortunately, black screen and `ERROR: openbox-xdg-autostart requires PyXDG to be installed` in the logs

blackadar, 2025-02-25:
Ah - the latest version works when I remove my mapping of `/dev/dri/renderD128`:

```
    devices:
      - /dev/dri/renderD128:/dev/dri/renderD128
```

This works on container version `v2.1.1-ls51` though..

blackadar, 2025-04-03:
Still seeing this issue with the latest image. It works fine until renderD128 is passed in. Anyone have an idea why?

blackadar, 2025-04-04:
So after lots of debugging.. turns out the issue was having only `/dev/dri/renderD128` passed in. When I swapped to:

```
    devices:
      - /dev/dri:/dev/dri
```

Everything worked without a problem. If anyone else is experiencing this issue - give that a try! Closing this since the docs (at least now) say to use /dev/dri and not just renderD128 - so this was just a config issue.

closed 2025-04-04 by blackadar