linuxserver/docker-orcaslicer #15: [FEAT] Expose latest releasa or beta versions on image tags

Pastitas, 2025-02-24:
### Is this a new feature request?

- [x] I have searched the existing issues

### Wanted change

To have a unstable or beta tag for the latest beta release

### Reason for change

I've got the need to use the beta version of orcaslicer, and I cannot find an appropiate tag on the builds for this

Pastitas, 2025-02-24:
One proposed change is to change in the dockerfile the url for downloads so that it's not hardcoded and any release tag can be accesed, like this 
```
curl -sX GET "https://api.github.com/repos/SoftFever/OrcaSlicer/releases/tags/${ORCASLICER_VERSION}" | awk '/browser_download_url/ && /AppImage/ {print $NF;exit;}' FS='[""]'
```

kitchung, 2025-03-11:
OrcaSlicer is a very active project and very frequently issue beta and release candidates that enthusiasts like us want to use. Please implement!

LinuxServer-CI, 2025-04-11:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

closed 2025-06-11 by LinuxServer-CI