linuxserver/docker-orcaslicer #17: [FEAT]  Location for "stl's" for docker space

WarHawk8080, 2025-02-28:
### Is this a new feature request?

- [x] I have searched the existing issues

### Wanted change

Can we get an option to point to a static location on the local machine to get access to .stl or other files in docker env please.  

### Reason for change

I run this on a networked computer in order to slice remotely when I am not on my own computer

LinuxServer-CI, 2025-03-30:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

Sboshoff76, 2025-04-24:
Yes please! A browser popup pointing to local file location would be tops.

j0nnymoe, 2025-04-24:
You can upload files to the container via the file manager:

[image]

Or what I personally do is map an additional volume to the container for where I store all my stl's.

Sboshoff76, 2025-04-25:
> You can upload files to the container via the file manager:
> 
> [image]
> 
> Or what I personally do is map an additional volume to the container for where I store all my stl's.

Sweet... Upload button FTW!
[image]

closed 2025-04-30 by j0nnymoe