linuxserver/docker-orcaslicer #19: [BUG] No build plate with nvidia GPU

zer0ish, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

When launching Orcaslicer docker container **with** an nvidia GPU, I get no build plate:

[image]

When launching Orcaslicer docker container **without** an nvidia GPU, I get a build plate:

[image]

Orcaslicer in the docker using a browser without the GPU acceleration makes it unusable when a model is loaded due to the laggy stuttering.

I've tried running the container as the App user (for TrueNAS) and root. Neither worked with GPU.

### Expected Behavior

Build plate to show up with the GPU variables shown in the docker hub page.

### Steps To Reproduce

Create the docker container with nvidia gpu as per the instructions on docker hub page and load the URL.

Remove the GPU variables and it loads the build plate.

### Environment

```markdown
- OS: TrueNAS 25.04.1
- How docker service was installed: Using Portainer template I created.
```

### CPU architecture

x86-64

### Docker creation

```bash
---
services:
  orcaslicer:
    image: lscr.io/linuxserver/orcaslicer:latest
    runtime: nvidia
    hostname: appname
    network_mode: bridge
    
    ports:
    - 2000:3000
    - 2001:3001

    restart: unless-stopped    
    
    environment:
        TZ: America/Toronto
        PUID: 568
        PGID: 568
        NVIDIA_VISIBLE_DEVICES: all
        NVIDIA_DRIVER_CAPABILITIES: all
        
    volumes:
    - /mnt/SASPool/portainer/orcaslicer_stack/config:/config
```

### Container logs

```bash
[... 215 lines cut ...]
(orca-slicer:329): GLib-GObject-CRITICAL **: 17:07:56.805: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.805: invalid (NULL) pointer instance
(orca-slicer:329): GLib-GObject-CRITICAL **: 17:07:56.805: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.805: invalid (NULL) pointer instance
(orca-slicer:329): GLib-GObject-CRITICAL **: 17:07:56.805: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.805: gtk_widget_add_events: assertion 'GTK_IS_WIDGET (widget)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.805: invalid (NULL) pointer instance
(orca-slicer:329): GLib-GObject-CRITICAL **: 17:07:56.805: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.856: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.856: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.857: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.857: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.870: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.870: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.872: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.872: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.874: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.874: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.882: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.882: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
(orca-slicer:329): GLib-GObject-WARNING **: 17:07:56.883: invalid cast from 'wxPizza' to 'GtkCellLayout'
(orca-slicer:329): Gtk-CRITICAL **: 17:07:56.883: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
 2025-05-28 17:08:02,740 [INFO] websocket 0: got client connection from 127.0.0.1
 2025-05-28 17:08:02,767 [PRIO] Connections: accepted: @10.77.77.25_1748466482.741095::websocket
```

LinuxServer-CI, 2025-06-28:
This issue has been automatically marked as stale because it has not had recent activity. This might be due to missing feedback from OP. It will be closed if no further activity occurs. Thank you for your contributions.

LinuxServer-CI, 2025-09-27:
This issue is locked due to inactivity